ogecko/alpaca-benro-polaris #23: Docker build process is not working

joriws, 2024-11-14:
**Describe the bug**
Multiple Docker issues in docker/Dockerfile
- ./requirements.txt is not found => ./docker/requirements.txt fixes 
- ./docker/entrypointALPACA.sh is not found => cp docker/entrypoint.sh docker/entrypointALPACA.sh
- after image build entrypoint.sh is not executable
- after it is executable there is no driver-folder - did you forget to git clone or copy the driver-part to image, 

**To Reproduce**
Steps to reproduce the behavior:
1.  bash ./docker/run.sh -b -t "Europe/Helsinki"

**Expected behavior**
Well the polaris:latest docker image to be build and container launched.

**Platform (please complete the following information):**
 - OS: RaspberryPi, Docker-ce

**Additional context**
Add any other context about the problem here.

jdmorriso, 2024-11-14:
@joriws, thanks for the feedback on the Docker/Raspberry Pi issues. As you've probably guessed, these have yet to be completed or tested. I need some new equipment and a gap in another project before I can work on them. 

If you have time and interest in helping with the project, let me know, and I can add you as a member and give you write access to create a branch for a pull request into the next version. If you can wait, I will probably address them before the next version, 1.0.1, is released.

joriws, 2024-11-16:
Thank you for you invititation to co-operate. For sure I can help you testing Raspberry/Docker installation but I am not experienced github user to manage pull-requests etc. The fixes I used to get Docker-version running are very trivial and maybe not the way you want to build packages and their folder structure, like 4th bullet copying ../../driver/* to ./docker/driver/ so making a copy of stuff as symlink did not do it.

Let me also use this opportunity writing to you to introduce my dream. It would be building Benro firmware -packet which could be installed directly to Polaris, and would directly include Alpaca-driver inside the mount without external hardware. The device is Linux and firmware-package is actually two ubifs-files providing boot/root and app folders. Not sure if there is more Benro-based checks for firmware than the MD5-hash + size at firmware config-file. I mean crypto-checksum with secret Benro key. 

This firmware is possible would allow also update libgphoto2 and gphoto2 packages, for some reason Benro's latest firmware package contained 2.5.27 version of packages when 2.5.31 has been released over year ago. 2.5.31 has quite many new camera additions, one which is on Polaris FB-group requested which I answered. Also improved support for my camera. 

Maybe I am just too afraid of testing that what if I brick my Polaris and no Benro support..

jdmorriso, 2024-12-15:
Made multiple corrections to the Dockerfile and scripts used to generate the Docker build.

closed 2024-12-15 by jdmorriso