ogecko/alpaca-benro-polaris #28: Failure to install on Rasp Pi

jlg84, 2025-06-04:
I am trying to install the alpaca controller for my Benro Polaris on a Raspberry Pi but the installation process fails consistently. I have tried to install it on fresh installations on a Rasp Pi 4 and a Rasp Pi 5, both with the Lite RaspPi OS and the full version, but right after the first installation process (curl -s https://raw.githubusercontent.com/ogecko/alpaca-benro-polaris/refs/heads/main/platforms/raspberry_pi/setup.sh | bash) I get errors as soon as it starts to instal Python 3.12.5:

Downloading Python-3.12.5.tar.xz...
-> https://www.python.org/ftp/python/3.12.5/Python-3.12.5.tar.xz
Installing Python-3.12.5...
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/jlg84/.pyenv/versions/3.12.5/lib/python3.12/bz2.py", line 17, in <module>
    from _bz2 import BZ2Compressor, BZ2Decompressor
ModuleNotFoundError: No module named '_bz2'
WARNING: The Python bz2 extension was not compiled. Missing the bzip2 lib?
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/jlg84/.pyenv/versions/3.12.5/lib/python3.12/curses/__init__.py", line 13, in <module>
    from _curses import *
ModuleNotFoundError: No module named '_curses'
WARNING: The Python curses extension was not compiled. Missing the ncurses lib?
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/jlg84/.pyenv/versions/3.12.5/lib/python3.12/ctypes/__init__.py", line 8, in <module>
    from _ctypes import Union, Structure, Array
ModuleNotFoundError: No module named '_ctypes'
WARNING: The Python ctypes extension was not compiled. Missing the libffi lib?
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ModuleNotFoundError: No module named 'readline'
WARNING: The Python readline extension was not compiled. Missing the GNU readline lib?
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/jlg84/.pyenv/versions/3.12.5/lib/python3.12/ssl.py", line 100, in <module>
    import _ssl             # if we can't import it, let the error propagate
    ^^^^^^^^^^^
ModuleNotFoundError: No module named '_ssl'
ERROR: The Python ssl extension was not compiled. Missing the OpenSSL lib?

Please consult to the Wiki page to fix the problem.
https://github.com/pyenv/pyenv/wiki/Common-build-problems

BUILD FAILED (Debian 12 using python-build 2.6.0)

Inspect or clean up the working tree at /tmp/python-build.20250604134008.2189
Results logged to /tmp/python-build.20250604134008.2189.log

Last 10 log lines:
	LD_LIBRARY_PATH=/tmp/python-build.20250604134008.2189/Python-3.12.5 ./python -E -m ensurepip \
		$ensurepip --root=/ ; \
fi
WARNING: Disabling truststore since ssl support is missing
Looking in links: /tmp/tmpmq9496pa
Processing /tmp/tmpmq9496pa/pip-24.2-py3-none-any.whl
Installing collected packages: pip
  WARNING: The scripts pip3 and pip3.12 are installed in '/home/jlg84/.pyenv/versions/3.12.5/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
Successfully installed pip-24.2
pyenv-virtualenv: `3.12.5' is not installed in pyenv.
Run `pyenv install 3.12.5' to install it.
pyenv: version `ssc-3.12.5' not installed
==SETUP== 4. Install the python dependencies needed for the application.
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    For more information visit http://rptl.io/venv

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
==SETUP== 5. Set up [systemd] services to start the polaris.service at boot time.
bash: line 47: cd: platformms/raspberry_pi: No such file or directory
cat: systemd/polaris.service: No such file or directory
==SETUP== 6. Starts the service.
Failed to enable unit: Unit file /etc/systemd/system/polaris.service is masked.
Failed to start polaris.service: Unit polaris.service is masked.
|-------------------------------------|
| alpaca-benro-polaris Setup Complete |
|                                     |
| You can access SSC via:             |
| http://polaris.local:5432       |
|                                     |
| Device logs can be found in         |
|  ./alpaca-benro-polaris/logs        |
|                                     |
| Systemd logs can be viewed via      |
| journalctl -u polaris               |
|                                     |
| Current status can be viewed via    |
| systemctl status polaris            |
|-------------------------------------|

Then if I try to do any of the further steps in the installation instructions I get errors about non-existing files (including the update.sh file to update the installed software).

jlg84, 2025-06-04:
That didn't get me any further along, I'm afraid.

blackmole, 2025-08-07:
Hey @jlg84,
For me the script also did not work, but I was able to install it manually.

`git clone https://github.com/ogecko/alpaca-benro-polaris.git
cd alpaca-benro-polaris
python -m venv venv
source venv/bin/activate
pip install -r platform/raspberry_pi/requirements.txt`

You can then start it with `python driver/main.py`. Make sure the WiFi of the RasPi is connected to the Polaris before that.
To install it as a service, I edited platform/raspberry_pi/systemd/polaris.service. I changed the WorkingDirectory and changed to ExecStart line to:
`ExecStart=/home/blackmole/alpaca-benro-polaris/venv/bin/python3 /home/blackmole/alpaca-benro-polaris/driver/main.py`
Then simply
`sudo cp platforms/raspberry_pi/systemd/polaris.service /etc/systemd/system
sudo systemctl enable polaris
sudo systemctl start polaris`

vyskocil, 2025-11-22:
For me the script also did not work, then I setup it manually : 

- I installed the Debian Trixie image from the Raspberry foundation (2025-10-01-raspios-trixie-arm64-full.img) on my Raspberry Pi 4 8go ([https://www.raspberrypi.com/software/operating-systems/](https://www.raspberrypi.com/software/operating-systems/)).
- I activated Bluetooth using the desktop icon
- I plugged a Realtek WIFI mini dongle, it's using the rtl8192cu linux module -> I wasted a lot of time trying to use this WIFI interface to connect to the Polaris network, it wasn't properly working. Then I switched the interface roles, using the pi4 internal WIFI to reach the Polaris and the Realtek dongle to connect to home WIFI
- installed the ABP 2.0 RC 1 sources in the home directory; then `cd alpaca-benro-polaris/platforms/raspberry_pi`
- I created a new Python env in ~/.pyenv using the system installed python version
```
python -m venv ~/.pyenv
export PATH=~/.pyenv/bin:$PATH
```
- I tried to install the needed module but it didn't worked, I had to install 2 Debian package first (yes Fortran language !!!):
```
sudo apt install gfortran
sudo apt install libopenblas-dev
```
- Then I could install the required python modules (and it took a very long time because it has to compile numpy and scipy locally on the SDCard which is very slow...):
```
pip install -r requirements.txt
```
- The config file need a tweak to start the ABP web server for the interface on port 8080 instead of 80:
```
vi config.toml
```
and change ` alpaca_pilot_port = 80` to `alpaca_pilot_port = 8080`
- Finally I was able to launch ABP 2.0 on the Raspberry !
```
python main.py
```
 
Then I was able to connect the ABP Pilot interface from another computer in the local network using `http://raspberryname.local:8080` !

closed 2025-12-09 by jdmorriso