ogecko/alpaca-benro-polaris #34: Better panoramas

wbuchanan, 2025-09-13:
**Is your feature request related to a problem? Please describe.**
As a Polaris user, I would like to have each panel in a panorama shot with the same horizon line so I can save time and effort post-processing later.

**Describe the solution you'd like**
I'm not sure if rotation around the RA axis is exposed in some way, but if it is I imagine it wouldn't be too difficult to compute a rotation angle based on the duration of all exposures taken for a panel, time between exposures, and the time to slew to the next position to send as a signal to rotate about the RA axis in the opposite direction.  This would make stacked and tracked panoramas more similar to how everyone would do it using other star trackers.

**Describe alternatives you've considered**
I've not tested if it is possible to accomplish this in NINA, but I assume it isn't a feature they would support given the primary users of that software are likely doing deep sky imaging nearly exclusively.

**Additional context**
When the camera continues rotating throughout the duration of a nightscape pano it can create artifacts when stitching the images together on the backend.

jdmorriso, 2025-09-13:
Version 2.0 of the driver will include an ASCOM rotator, enhancing control over the camera's roll position and its equatorial position angle. This improvement should facilitate the creation of DSO panoramas by maintaining a consistent equatorial position angle across different panorama coordinates. It should also help when re-centering to correct for drift.

The current Benro Polaris firmware does not calculate or manage the equatorial position angle. Whenever a goto is performed, it resets the roll angle to zero and turns tracking on, which increases the roll angle in a consistent manner with sidereal tracking. It currently has no concept of equatorial position angle.

For non-DSO work, doesn't the current firmware function as you describe? Every time it moves to a new coordinate, the camera frame is made horizontal, or the roll position is set back to zero. If tracking is off, it should not rotate the frame. Please let me know if I'm not understanding your request correctly.

wbuchanan, 2025-09-13:
@jdmorriso when using the Polaris for stacked and tracked panoramas, the camera continues tracking the entire duration from the first to last sub.  When trying to shoot 360 degree panoramas with 4 x 1' subs and a wide angle lens, the horizon is pretty far from level by the time it gets to the last panel.  If shooting RGB followed by using an Ha filter things are even worse.  I've not attempted this using NINA, but this is the behavior when using the polaris alone.

jdmorriso, 2025-09-13:
ok, thanks, I better understand the issue now. When I am completing the rotator implementation, I'll see whether this can be addressed.

jdmorriso, 2025-10-15:
Note to self - this may be fixed with Nina's Preserve Alignment option on the Framing Tab. When you select more than 1 panel for a mosaic you can choose whether the rotation angle is adjusted per shot to get a rectangular panorama. This might fix the issue with the stock Polaris firmware when parnoramas go up higher in the sky and overlap way more near Zenith too.

jdmorriso, 2025-11-05:
This has been tested in Nina's Preserve Alignment option.